isaac-sim/IsaacGymEnvs #208: Train with SAC

drsssssss, 2024-03-12:
Hello! I see that the project contains **isaacgymenvs/cfg/task/AntSAC.yaml**, and I'm wondering if I can use SAC for training? thanks!!!

Yougo-robotics, 2024-05-28:
Yes you can you will find some config file for SAC here : 
https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/tree/main/isaacgymenvs/cfg/train
By default IsaacGymEnvs is using rlgames for the rl part but you can easily use other libraries that support parallel envs.

marsxw, 2025-02-16:
`python train.py task=Ant  train=AntSAC headless=True `

An error occurred 

Error executing job with overrides: ['task=Ant', 'train=AntSAC', 'headless=True']
Traceback (most recent call last):
  File "train.py", line 210, in launch_rlg_hydra
    runner.run({
  File "/root/anaconda3/envs/rlgpu/lib/python3.8/site-packages/rl_games/torch_runner.py", line 133, in run
    self.run_train(args)
  File "/root/anaconda3/envs/rlgpu/lib/python3.8/site-packages/rl_games/torch_runner.py", line 116, in run_train
    agent.train()
  File "/root/anaconda3/envs/rlgpu/lib/python3.8/site-packages/rl_games/algos_torch/sac_agent.py", line 541, in train
    step_time, play_time, update_time, epoch_total_time, actor_losses, entropies, alphas, alpha_losses, critic1_losses, critic2_losses = self.train_epoch()
  File "/root/anaconda3/envs/rlgpu/lib/python3.8/site-packages/rl_games/algos_torch/sac_agent.py", line 529, in train_epoch
    return self.play_steps(random_exploration)
  File "/root/anaconda3/envs/rlgpu/lib/python3.8/site-packages/rl_games/algos_torch/sac_agent.py", line 497, in play_steps
    self.obs = next_obs.clone()
AttributeError: 'dict' object has no attribute 'clone'