isaac-sim/IsaacGymEnvs #220: Possible with H100 GPU?

LyuJZ, 2024-05-15:
Hi, I am going to use H100 GPU with isaacgym. However, the code will stop after executing this line https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/blob/aeed298638a1f7b5421b38f5f3cc2d1079b6d9c3/isaacgymenvs/tasks/base/vec_task.py#L63 and evoke "segmentation fault" in the terminal. Does anyone have similar issues?

ankile, 2024-05-20:
Yes, I have the exact same issue, i.e., when loading the sim, it just prints a segmentation fault after it prints `+++ Using GPU PhysX`, with no other error messages at all as far as I can tell... The same environment runs on Nvidia 3090, A6000, and V100s...

Any updates from your side?

edwardjjj, 2024-09-12:
@ankile Hi, just wondering if you found a solution to this.

ankile, 2024-09-12:
@edwardjjj no, unfortunately, I haven't... It's quite frustrating that there's no support and that the successor library, IsaacSim/Lab explicitly does not support any of the datacenter GPUs as RT (raytracing) cores are required at the same time as competitors like Mujoco (MJX) and Sapien support any GPU out of the box... I hope they get this sorted soon as the Isaac platform seemed very promising.

edwardjjj, 2024-09-13:
> @edwardjjj no, unfortunately, I haven't... It's quite frustrating that there's no support and that the successor library, IsaacSim/Lab explicitly does not support any of the datacenter GPUs as RT [...]

Thanks for the quick response. It is unfortunate that even the latest IsaacSim does not support high-end datacenter GPUs. It is frustrating that we are not able to utilize the latest GPUs for cutting edge research... Hope they manage to add support soon.

edwardjjj, 2024-09-14:
@ankile Hi, just want to share this info with you. I talked to an IsaacSim dev on their discord server. He confirmed that it is impossible to support these data center GPUs because they don't have the RTX chips. The best cards out there that have RTX chips are the a6000 and L40s.

MiangChen, 2024-11-21:
https://github.com/isaac-sim/IsaacGymEnvs/issues/52#issuecomment-1299586926
is this helpful? 
change the nvidia driver versiton to 510.85.02 and CUDA version to 11.6

OAHC2022, 2025-07-24:
Hi, just wanted to follow up to check if IsaacGym now supports the H100 GPU. Thanks!