isaac-sim/IsaacGymEnvs #221: Question about running on Ubuntu22.04

RobinWitch, 2024-05-28:
Hello,

I am trying to use this on Ubuntu 22.04 via VNC(TigerVNC) with a 4090 GPU. However, When I run 'python examples/joint_monkey.py', I am encountering a strange problem:
`
Importing module 'gym_38' (/mnt/data/cbh/Eureka/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /mnt/data/cbh/Eureka/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
WARNING: Forcing CPU pipeline.
Not connected to PVD
+++ Using GPU PhysX
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: disabled
Xlib: extension "GLX" missing on display ":1".
Xlib: extension "GLX" missing on display ":1".
Xlib: extension "GLX" missing on display ":1".
Xlib: extension "GLX" missing on display ":1".
Xlib: extension "GLX" missing on display ":1".
Xlib: extension "GLX" missing on display ":1".
Segmentation fault (core dumped)
`
Interestingly, I tested this on another server with Ubuntu 20.04 and a 3090 GPU using the same VNC server program((TigerVNC)), and it worked successfully. Both computers are cloud servers. I wonder how I can solve this problem.

Any help would be greatly appreciated.

RobinWitch, 2024-05-30:
Sorry, it's my problem. I have solved.

closed 2024-05-30 by RobinWitch

TengBoYuu, 2024-07-15:
Hi, I have the same problem, how did you fix it. Thx☺️

RobinWitch, 2024-07-15:
I use vnc to visualize screen on remote server, in fact I mistake my vncserver installed in remote server, I use TightVNC instead of TigerVNC.

I uninstall TightVNC and install TigerVNC, the error then has been solved.

> Hi, I have the same problem, how did you fix it. Thx☺️

TengBoYuu, 2024-07-15:
> I use vnc to visualize screen on remote server, in fact I mistake my vncserver installed in remote server, I use TightVNC instead of TigerVNC.
> 
> I uninstall TightVNC and install TigerVNC, the error then has been solved.
> 
> > Hi, I have the same problem, how did you fix it. Thx☺️

Thx, I run my program on TurboVNC, it doesn't have GPU acceleration, but the Isaay Gym maybe neet it, I'll try TigerVNC.