isaac-sim/IsaacGymEnvs #229: Up to cuda 11.8 worked environment

louhz, 2024-08-02:
some trick for use isaacgym in torch 2.0
Change the rlgpu_conda_env.yml to this version and it may work(at least for my application) 

name: issacgym
channels:
  - pytorch
  - conda-forge
  - defaults
dependencies:
  - python=3.8
  - pytorch>=1.8.1
  - torchvision>=0.9.1
  - cudatoolkit>=11.1
  - pyyaml>=5.3.1
  - scipy>=1.5.0
  - tensorboard>=2.2.1
  - mkl==2024.0

here is my worked key conda  environment:
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                  2_kmp_llvm    conda-forge
absl-py                   2.1.0              pyhd8ed1ab_0    conda-forge
aom                       3.5.0                h27087fc_0    conda-forge
blas                      2.121                       mkl    conda-forge
blas-devel                3.9.0            21_linux64_mkl    conda-forge
brotli-python             1.1.0            py38h17151c0_1    conda-forge
bzip2                     1.0.8                h4bc722e_7    conda-forge
c-ares                    1.32.3               h4bc722e_0    conda-forge
ca-certificates           2024.7.4             hbcca054_0    conda-forge
certifi                   2024.7.4           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0           py38h6d47a40_0    conda-forge
charset-normalizer        3.3.2              pyhd8ed1ab_0    conda-forge
cudatoolkit               11.8.0              h4ba93d1_13    conda-forge
expat                     2.6.2                h59595ed_0    conda-forge
ffmpeg                    5.1.2           gpl_h8dda1f0_106    conda-forge
filelock                  3.15.4             pyhd8ed1ab_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 h77eed37_2    conda-forge
fontconfig                2.14.2               h14ed4e7_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freetype                  2.12.1               h267a509_2    conda-forge
gettext                   0.22.5               h59595ed_2    conda-forge
gettext-tools             0.22.5               h59595ed_2    conda-forge
gmp                       6.3.0                hac33072_2    conda-forge
gmpy2                     2.1.5            py38h6a1700d_1    conda-forge
gnutls                    3.7.9                hb077bed_0    conda-forge
grpcio                    1.62.2           py38h94a1851_0    conda-forge
h2                        4.1.0              pyhd8ed1ab_0    conda-forge
hpack                     4.0.0              pyh9f0ad1d_0    conda-forge
hyperframe                6.0.1              pyhd8ed1ab_0    conda-forge
icu                       75.1                 he02047a_0    conda-forge
idna                      3.7                pyhd8ed1ab_0    conda-forge
importlib-metadata        8.2.0              pyha770c72_0    conda-forge
jinja2                    3.1.4              pyhd8ed1ab_0    conda-forge
jpeg                      9e                   h0b41bf4_3    conda-forge
lame                      3.100             h166bdaf_1003    conda-forge
lcms2                     2.15                 hfd0df8a_0    conda-forge
ld_impl_linux-64          2.40                 hf3520f5_7    conda-forge
lerc                      4.0.0                h27087fc_0    conda-forge
libabseil                 20240116.2      cxx17_he02047a_1    conda-forge
libasprintf               0.22.5               h661eb56_2    conda-forge
libasprintf-devel         0.22.5               h661eb56_2    conda-forge
libblas                   3.9.0            21_linux64_mkl    conda-forge
libcblas                  3.9.0            21_linux64_mkl    conda-forge
libdeflate                1.17                 h0b41bf4_0    conda-forge
libdrm                    2.4.122              h4ab18f5_0    conda-forge
libexpat                  2.6.2                h59595ed_0    conda-forge
[... 70 lines cut ...]
tbb                       2021.12.0            h434a139_3    conda-forge
tensorboard               2.17.0             pyhd8ed1ab_0    conda-forge
tensorboard-data-server   0.7.0            py38hcdda232_1    conda-forge
tk                        8.6.13          noxft_h4845f30_101    conda-forge
torchvision               0.15.2                 py38_cpu    pytorch
typing_extensions         4.12.2             pyha770c72_0    conda-forge
urllib3                   2.2.2              pyhd8ed1ab_1    conda-forge
werkzeug                  3.0.3              pyhd8ed1ab_0    conda-forge
wheel                     0.43.0             pyhd8ed1ab_1    conda-forge
x264                      1!164.3095           h166bdaf_2    conda-forge
x265                      3.5                  h924138e_3    conda-forge
xorg-fixesproto           5.0               h7f98852_1002    conda-forge
xorg-kbproto              1.0.7             h7f98852_1002    conda-forge
xorg-libx11               1.8.4                h0b41bf4_0    conda-forge
xorg-libxau               1.0.11               hd590300_0    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h0b41bf4_2    conda-forge
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-xextproto            7.3.0             h0b41bf4_1003    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.6                h166bdaf_0    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
zipp                      3.19.2             pyhd8ed1ab_0    conda-forge
zstandard                 0.23.0           py38h62bed22_0    conda-forge
zstd                      1.5.6                ha6fb4c9_0    conda-forge

ShallowDreamKenny, 2024-10-23:
When i update the yaml file, run the test in isaacgymenvs. A new error occurs "RuntimeError: Error building extension 'gymtorch'"

louhz, 2024-10-28:
> When i update the yaml file, run the test in isaacgymenvs. A new error occurs "RuntimeError: Error building extension 'gymtorch'"

I remember this is due to the incorrect file path >>