isaac-sim/IsaacGymEnvs #247: Tacsl branch trained error

Junyu-Zhu, 2024-12-17:
I've followed this `isaacgymenvs/tacsl_sensors/install/tacsl_setup.md` and establish the conda environment, but when I tried policy learning with this step
`cd isaacgymenvs/tacsl_sensors/install`
`./train_state.sh`
It appears `return torch.normal(self.loc.expand(shape), self.scale.expand(shape))`
`RuntimeError: normal expects all elements of std >= 0.0`
How should I fix this error, thanks.

molyswu, 2025-05-08:
the same problem   ：
RuntimeError: normal expects all elements of std >= 0.0

wusiyu66, 2025-09-13:
> the same problem ： RuntimeError: normal expects all elements of std >= 0.0

TacSL-Specific Isaac Gym binary is locked behind a restricted Google drive .I see you are already using it.Could you help me to offer your version？