isaac-sim/IsaacGymEnvs #258: Does 50 series work？

MyGospelzz, 2025-03-26:
I want to buy a net card like 5070ti but I have some some on it,I wanna know it can work.

tony-hj, 2025-04-15:
I bought a 5070ti graphics card, but its computing architecture is sm_120. Currently, pytorch and related packages that support this architecture require a minimum python3.9 version, but isaac gym only supports python3.8 at most. I tried to change the version requirement in the setup.py file in isaacgym to <3.10 to force isaacgym to be installed in python3.9. The installation was successful, but when running isaacgym, an error occurred that the models could not be read because they were all compiled in python3.8. I don’t know how to solve this adaptation problem now. My 5070ti can’t continue to train my previous work. I’m very anxious. I hope the official can have a solution! ! !

tony-hj, 2025-04-15:
@ViktorM @alex-petrenko @kellyguo11  
Dear friends, I hope to get your help, PLEASE!!!!

Yonghao-Yu, 2025-04-15:
> I bought a 5070ti graphics card, but its computing architecture is sm_120. Currently, pytorch and related packages that support this architecture require a minimum python3.9 version, but isaac gym [...]

same question

Hahalim2022y, 2025-07-07:
> I bought a 5070ti graphics card, but its computing architecture is sm_120. Currently, pytorch and related packages that support this architecture require a minimum python3.9 version, but isaac gym only supports python3.8 at most. I tried to change the version requirement in the setup.py file in isaacgym to <3.10 to force isaacgym to be installed in python3.9. The installation was successful, but when running isaacgym, an error occurred that the models could not be read because they were all compiled in python3.8. I don’t know how to solve this adaptation problem now. My 5070ti can’t continue to train my previous work. I’m very anxious. I hope the official can have a solution! ! !我买的是5070 ti显卡，但它的计算架构是sm_120。目前，支持这种架构的pytorch和相关软件包最低需要python3.9版本，但isaac gym最多只支持python3.8。我尝试将isaacgym中setup.py文件中的版本要求更改为 3.10，以强制isaacgym安装在python3.9中。安装是成功的，但是当运行isaacgym时，出现了一个错误，无法读取模型，因为它们都是在python3.8中编译的。我现在不知道如何解决这个适应问题。我的5070 ti不能继续训练我以前的工作。我很焦虑。希望官方能有办法！! !

same question